secretkeysio/GCAdapterDriver #1: Where is the GCAdapterDriver.kext file? Do I rename one of the files or something? HELP PLZ :(

sfay1996, 2021-04-15:
One of the instructions is to 

Copy GCAdapterDriver.kext to /Library/Extensions/GCAdapterDriver.kext

This zip doesnt have a file called GCAdapterDriver.kext in it. Theres no instructions on where to find it, it just says to copy it to extensions.

There is a kext folder with another folder called GCAdapterDriver. I then copied this to /library/extensions and then added .kext to the file name and attempted the steps after.... No dice.. 

Im sure someone out there has to have had this problem before. i.e overall failure at setting it up period Lol. Do you know any way to help or a youtube instructional video on setting this MacOS overclock up? Thanks so much, i just wanna get off this 125 hz BS lmao 

Thanks so much :)

ryanmcgrath, 2021-04-15:
Click Assets on the release page and download the correct one for your architecture. It’s hidden by GitHub for whatever reason.

What macOS version are you on?

sfay1996, 2021-04-15:
I am on High Sierra 10.13.6, i dont know what you mean by assets on the release page. I just went to the code tab and hit download zip. Im sorry i dont understand :(

sfay1996, 2021-04-15:
Also thanks very much for getting back to me so quickly!

ryanmcgrath, 2021-04-15:
Hmmm, this is more user help than project issue (but I can help, sure). I'm going to close this, _but_ go ahead and email me (ryan@rymc.io) and we can go over things there - or if you happen to be in the Slippi Discord I'm around there too.

closed 2021-04-15 by ryanmcgrath

sfay1996, 2021-04-15:
awesome, I am in the discord. Il message you there. Thanks again!