secretkeysio/GCAdapterDriver #3: Error Opening Adapter: Access denied (insufficient permissions)

RyanVillenaUCR, 2021-11-17:
Getting this error upon hitting "Configure" in Dolphin. Running GCAdapterDriver v1.6 (the release for MacOS 11.0+), Dolphin 5.0-15445, and Monterey 12.0.1.

[image]

Looks like I'm not the first this issue has happened to, as [this issue](https://github.com/secretkeysio/GCAdapterDriver/issues/2#issue-1050515391) just last week documents. Unfortunately it was closed without explanation, so I'm reopening one here hoping for a more definitive solution.

I'm also wondering if someone can confirm if [temporarily disabling System Integrity Protection (SIP)](https://winerrorfixer.com/error-opening-adapter-access-denied/#Method_1_Temporarily_Disabling_the_System_Integrity_Protection_SIP) is a valid solution; I'm unfortunately not able to test this on my Mac for security reasons.

ryanmcgrath, 2021-11-17:
No, you do not need to disable SIP - these drivers are signed so that you _don't_ have to do that.

Usually when this happens it takes some combination of unplugging and replugging the device after the driver is activated; as far as I understand it this is due to a bug ("feature") in DriverKit where Apple's drivers don't automatically detach. Giving this a few tries usually clears it up for people, sometimes with a restart of Dolphin in there.

(Also, if you're sure you want to use Dolphin 5.0 then by all means go ahead - but I think you probably want a nightly, as the stable release is fairly old by this point)

RyanVillenaUCR, 2021-11-17:
It ended up being an accurate error message; the driver needed permissions from System Preferences that it wasn't vocal about asking for. I found this by noticing the small [image] icon in the top toolbar. I believe it needed keystroke-writable permissions, and one other one that I can't remember. After doing this and unplugging/replugging like you said, it worked for me. I appreciate the help!

Not sure if it's my place to do so here, but it might be a good idea to add this to the FAQ on the distribution site if it can steer more users in the right direction. For now I'm closing this issue.

closed 2021-11-17 by RyanVillenaUCR

ryanmcgrath, 2021-11-17:
Hmmm, when you first activated the driver, did you click the prompt to go into system preferences and allow the driver? If you didn't, it's possible it was that (System Preferences > Security & Privacy > Bottom would show an `enable` button).

The driver shouldn't need keystroke-writable or anything, as it touches nothing with that...

ryanmcgrath, 2021-11-17:
Rather, I should ask: had you run the app but not clicked the menubar icon? The toggle switch in there has to be switched on before the driver is actually loaded, and the OS will auto-prompt for any permissions it needs with a modal dialog.

The site says this:

>Start the app (it runs in your menubar). Click the icon, then toggle the driver.

I suppose I can find time to shove a screenshot on there if it's confusing, but lemme know if my understanding is clear or if I'm missing something. Glad it works now!

RyanVillenaUCR, 2021-11-17:
> Rather, I should ask: had you run the app but not clicked the menubar icon? 
> 
> The site says this:
> 
> > Start the app (it runs in your menubar). Click the icon, then toggle the driver.

My fault, I was looking for the name of the driver to show up in text on the left side of the menu bar; I hadn't realized (or just skimmed past) that it'd have been an icon instead. Toggling that was what made it work.

I now honestly couldn't tell you if I had to enable permissions after that. I genuinely misremembered giving the driver Input Monitoring permissions - I'm checking now and it seems that in the timeframe I spent getting Slippi to work, it was Slippi Launcher and Dolphin that I'd given those permissions instead.

Sorry for wasting your time with this. But, really glad it works!! Thanks for all your efforts on this project.