secretkeysio/GCAdapterDriver #5: Adding support for RockBand USB drumkit

neomanic, 2021-12-09:
From my searching through the Dolphin wiki and from what I can glean from various forums, I understand that the DriverKit based driver would be the best way to add support for the Rock Band USB drumkit (which I have), and then possibly some other USB HID Wii peripherals (thinking of getting one of the keyboard controllers too).

From a quick look through the source, it looks to me like the GCN adapter product/vendor ids are in the Info.plist, and that's about the only thing that is specific to the GCN as opposed to other USB devices... so are you open to adding some of these other Wii peripherals?

ryanmcgrath, 2021-12-09:
I actually do have an update I have to push for the Skylanders Portal or whatever, so if people can provide me a list to just add... I'm not opposed. I just don't own these devices to test, is all, so it would be ideal if people could test on their own (even via a codeless kext, if they're comfortable).

Cain73, 2022-06-15:
Good day Ryan! I have a Skylanders Portal of Power and would be willing to test for any solution to get this to work on my Mac if given instructions. I'm only a user and am not that techsmart when it comes to emulation, but I will test whatever you need for this. It's a win win for me! Thank you so much. 

P.S I'm open to many forms of communication, messages, discord etc, just say the word.

deBaer, 2022-09-04:
I have the wireless keyboard receiver and I would love to use it. The USB VID is 1bad and the PID is 3330.

neomanic, 2022-09-05:
@ryanmcgrath apologies, I didn't see your reply to me at the time... I will test as soon as I next get the chance. What stopped me was the whole disabling SIP issue... but I'm pretty sure I should be able to test in a macOS VM.

I have since bought and started using the RockBand Wii Pro adapter which allows the use of drums (bought an e-kit!), keyboards and guitar, so I will test and provide VID/PIDs for that as well.

Kneudel, 2024-02-23:
> I actually do have an update I have to push for the Skylanders Portal or whatever, so if people can provide me a list to just add... I'm not opposed. I just don't own these devices to test, is all, [...]

Any News about the Skylanders Portal Support? For Windows and Linux there are workarounds to make it works with the Dolphin Emulator and fix the "blink issue"  There is currently no solution for macOS. 
[https://wiki.dolphin-emu.org/index.php?title=USB_Passthrough](url) say perhaps there will be a solution with the GCAdapterDriver.

ryanmcgrath, 2024-02-23:
IIRC Dolphin merged a software solution for this so you don't _need_ a real portal - or am I misremembering?

Kneudel, 2024-02-23:
> IIRC Dolphin merged a software solution for this so you don't _need_ a real portal - or am I misremembering?

Yes, your are right, but some hardcore fans would like to use their portals, and yeah if I plugged the portal to my Mac (user a ps4 portal for Skylanders Trap Team) it is recorgnized by the Mac and listed on the device list in the Dolphin USB Passthrough menu. If it is connected to the game, the game reacting to it like it is connected and disconnected in a loop. You can see it here.
https://youtube.com/clip/UgkxiMCKY4hyhJV05JfU5Ia8JqYnQ8cH1ub6?si=CDnghVF-KbQKIh8W
The workaround of this under Windows is Zadig and for Linux its /etc/udev/rules.d/52-dolphin.rules descripted here:
https://wiki.dolphin-emu.org/index.php?title=USB_Passthrough

sanjay900, 2025-01-31:
This was just built, see if it works https://secretkeys.io/gcadapterdriver/GCAdapterDriver-1.8-beta.app.zip