secretkeysio/GCAdapterDriver #6: m1

hermanbanken, 2022-01-18:
I tried installing the ARM kext, after setting "Reduced Security" at the Startup Options.

Then running ([src](https://developer.apple.com/documentation/apple-silicon/installing-a-custom-kernel-extension)):
```bash
kmutil load -p ~/Library/Extensions/GCAdapterDriver.kext
```

fails with 
```
Error Domain=KMErrorDomain Code=30 "Validating extension failed: Kext com.secretkeys.driver.GCAdapterDriver v1 in executable kext bundle com.secretkeys.driver.GCAdapterDriver at /private/var/db/KernelExtensionManagement/Staging/com.secretkeys.driver.GCAdapterDriver.ohkpcD/GCAdapterDriver.kext:

Filesystem error: Invalid ownership (501:20) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.secretkeys.driver.GCAdapterDriver.ohkpcD/GCAdapterDriver.kext/Contents
Filesystem error: Invalid ownership (501:20) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.secretkeys.driver.GCAdapterDriver.ohkpcD/GCAdapterDriver.kext/Contents/_CodeSignature
Filesystem error: Invalid ownership (501:20) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.secretkeys.driver.GCAdapterDriver.ohkpcD/GCAdapterDriver.kext/Contents/_CodeSignature/CodeResources
Filesystem error: Invalid ownership (501:20) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.secretkeys.driver.GCAdapterDriver.ohkpcD/GCAdapterDriver.kext/Contents/MacOS
Filesystem error: Invalid ownership (501:20) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.secretkeys.driver.GCAdapterDriver.ohkpcD/GCAdapterDriver.kext/Contents/MacOS/GCAdapterDriver
Filesystem error: Invalid ownership (501:20) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.secretkeys.driver.GCAdapterDriver.ohkpcD/GCAdapterDriver.kext/Contents/Info.plist
Authenticating extension failed: Bad code signature" UserInfo={NSLocalizedDescription=Validating extension failed: Kext com.secretkeys.driver.GCAdapterDriver v1 in executable kext bundle com.secretkeys.driver.GCAdapterDriver at /private/var/db/KernelExtensionManagement/Staging/com.secretkeys.driver.GCAdapterDriver.ohkpcD/GCAdapterDriver.kext:

Filesystem error: Invalid ownership (501:20) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.secretkeys.driver.GCAdapterDriver.ohkpcD/GCAdapterDriver.kext/Contents
Filesystem error: Invalid ownership (501:20) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.secretkeys.driver.GCAdapterDriver.ohkpcD/GCAdapterDriver.kext/Contents/_CodeSignature
Filesystem error: Invalid ownership (501:20) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.secretkeys.driver.GCAdapterDriver.ohkpcD/GCAdapterDriver.kext/Contents/_CodeSignature/CodeResources
Filesystem error: Invalid ownership (501:20) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.secretkeys.driver.GCAdapterDriver.ohkpcD/GCAdapterDriver.kext/Contents/MacOS
Filesystem error: Invalid ownership (501:20) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.secretkeys.driver.GCAdapterDriver.ohkpcD/GCAdapterDriver.kext/Contents/MacOS/GCAdapterDriver
Filesystem error: Invalid ownership (501:20) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.secretkeys.driver.GCAdapterDriver.ohkpcD/GCAdapterDriver.kext/Contents/Info.plist
Authenticating extension failed: Bad code signature}
```

Am I doing something wrong or is this kext not signed correctly? Why does DriverKit not work on M1? Simply no time to update it yet, or fundamentally impossible?

ryanmcgrath, 2022-01-18:
_DriverKit_ works on M1; you shouldn't be using a kext for that, you should be using GCAdapterDriver.app: https://secretkeys.io/gcadapterdriver/

If you're after _overclocking_, that is not currently supported on DriverKit (so Big Sur/Monterey) due to a bug in DriverKit that I reported to Apple some time ago regarding polling-rate changes not taking effect. Catalina _has_ DriverKit, however it has numerous other bugs and in general a kext is used there for sanity.

If you want to overclock on Big Sur or Monterey, there's a separately maintained guide over here: https://github.com/project-slippi/Ishiiruka/wiki/Overclocking-Controllers-on-macOS-Big-Sur-or-Monterey

If Apple does not fix the bug at some point here I will likely sign the kext for tinkerers who know what they're doing and can install themselves, but I don't expect there to be a signed PKG installer for it. I would absolutely prefer to just do it in DriverKit though and keep a clean separation of drivers here, as kexts are increasingly annoying to deal with.

Going to close this as it's not a bug (at least not with GCAdapterDriver).

closed 2022-01-18 by ryanmcgrath

hermanbanken, 2022-01-19:
I also tried installing the app. Unless my controller is DOA… I’m not sure it is working.

The releases text was a bit vague about whether M1 users could use the driverkit app. After rereading it I still thought that.

ryanmcgrath, 2022-01-19:
Eh? The website has M1 for DriverKit, top right:

[image]

On the releases page, I tweaked the verbage to make it clear it's solely a paragraph about M1 kext support. I figured it was originally clear as the paragraph contains:

>Note that kernel extensions are different

But regardless that should be updated. In _general_ I don't use the Releases page as anything other than notable code bumps - the website is the source for working products.

If you installed the `.app`, look for the icon in your menubar that looks like a port. It has a dropdown with a switch to activate the driver. The DMG download on the page should pop open a guide explaining this.

hermanbanken, 2022-01-20:
Sorry I feel stupid for kinda using Issues as support. I missed the M1 on the website. The fact it ain’t working is probably caused by my controller.

Thanks for your hard work and thanks for taking the time to move to the User Space (DriverKit) future! ❤️

ryanmcgrath, 2022-01-21:
Heh, not stupid - it's not exactly clear where to go to _get_ support.

Regardless, I hope the driver works for you. If you do find a bug or something definitely feel free to open another issue here. :)