secretkeysio/GCAdapterDriver #7: Driver Not Downloading

addisontuohey, 2022-02-07:
Hello! I've tried downloading this driver multiple times onto my 10.14.6 Macbook Pro. I keep searching for the driver in my systems/library/extensions folder, but despite running the installer multiple times with permission granted, the driver still has not appeared in my extensions folder. Not sure what is wrong! Please help if you are able.

ryanmcgrath, 2022-02-07:
- Did you run the installer `.pkg`?

- Are you running `sudo ls /Library/Extensions`?

addisontuohey, 2022-02-07:
To specify, it seems like the installer is not working.

addisontuohey, 2022-02-07:
Yup! I ran the installer and it says it's installed, but when I check the extensions folder there's unfortunately nothing there.

addisontuohey, 2022-02-07:
[image]
)

although it seems to maybe come up when I run sudo ls /Library/Extensions?

ryanmcgrath, 2022-02-07:
Glad it's working. :)

Drivers can be a bit funky sometimes, going to guess it was down to that. Enjoy!

closed 2022-02-07 by ryanmcgrath