secretkeysio/GCAdapterDriver #8: M1 Monterey 12.3.1 -- Controllers never connecting, despite Activation notice.

HomeStarRunnerTron, 2022-04-26:
Hi-hi-- thank you so much for making this real classy-lookin' driver. I haven't been able to get it to work yet on my MacBook Pro. I have both of the USB cords of my official Wii U Gamecube Adapter plugged into my computer through USB-C dongles (I dunno, does it matter if I plug in the black or grey cord first?), and I've got a nice new Gamecube controller plugged into Port 1, and nothin'. It doesn't show up on Dolphin, OpenEmu, or my controller mapper, etc.

I check my csrutil status, and it's disabled... my Security and Privacy settings about Apps is set to "Allow apps downloaded from: Anywhere"...

Every time I turn the slider for GCAdapterDriver off, it makes me type in my admin password...

[image]

[image]

In my SystemExtensions folder, I've got these three folders from my three attempts turning on the extension... lemme turn it back on.

[image]

Tah-dah! Another identical GCAdapterDriver dext in a folder!

[image]

_Seeee?_

[image]

And to show, it keeps trying to make me allow it over and over again, but I don't think it's actually working, even though the slider says Activated.

[image]

Let me know if there's anything I can do, all the very best! ^_^

ryanmcgrath, 2022-04-26:
Appreciate the detailed issue. :)

1) There's no need to disable SIP, and in fact disabling it on some Macs reduces the quality of things like streaming music or videos. You should consider turning it back on unless you have a reason to keep it off (the driver is signed to allow SIP to stay enabled).

2) What type of adapter are you using? If it's a Mayflash or adapter with a PC/Wii/WiiU mode switch, make sure it's in Wii/WiiU mode. And what model Mac is this?

3) Black cord is the important one. I assume you're unplugging + replugging the adapter (not a USB hub) from the Mac after activating, yeah?

HomeStarRunnerTron, 2022-04-26:
Woah, thank you so much for your prompt response, amazing!

1. Ahh, I keep it off for other programs, but I can see what happens if I turn it back on.
2a. I'm using the official Wii U GameCube adapter, which doesn't have any switches, I believe!
2b. This is a MacBook Pro 14-inch, 2021.
3. Yes, I am attempting to unplug and replug each time, but who knows if I've been messy in that, haha. Order of operations between each turn on and turn off can get a little odd.

ryanmcgrath, 2022-04-26:
Ah, no worries if you know you want SIP off then - the driver works with it disabled as well, I just tell people this since it's really not obvious and tutorials for drivers from older eras consistently say to disable it... which is less than ideal for most.

Kk, those devices should work just fine. What USB-C adapter are you using? If you plug in the device and then navigate to `Apple Icon > About This Mac > System Report > USB`, what's it show?

Re: order of operations... with DriverKit, the best bet is usually:

- Turn on driver via the switch
- Plug in black cord
- Unplug it... plug it back in...

There's a weird glitch where sometimes Apple's drivers will still latch on first connect, so the unplug + replug is necessary sometimes. Go figure.

HomeStarRunnerTron, 2022-04-26:
I'm using a Nimaso USB-C adapter which has worked for every drive or adapter I've plugged into it!

https://www.amazon.com/NIMASO-Adapter-Adaptor-Compatible-Pixel-Gray/dp/B07TJ38SDW

*plugs in the black cord and looks in System Report > USB* Aha, this is so useful, thank you!!

[image]

*uninstalls driver completely, deleting all those duplicate dexts and then tries to switch it on again, plugs in black cord, and unplugs it, and plugs it back in*

Nothin', but it still keeps showing up in System Report the same way, which at least means it's been read!

ryanmcgrath, 2022-04-26:
When you trash those dexts, you probably want to run this in a terminal instead...:

`sudo systemextensionsctl reset`

This removes the drivers properly... you may also need to enable developer mode to run the above:

`sudo systemextensionsctl developer on`

Otherwise the best way to remove a DriverKit driver is to trash the app, and then clear the trash... which is also non-obvious.

All this said, run this in a Terminal to see what driver is attaching to that USB device:

`ioreg | grep WUP-028 -A6`

ryanmcgrath, 2022-04-26:
(Restarting the machine also helps clear out old dexts, I believe)

HomeStarRunnerTron, 2022-04-26:
Donezo, the "reset" worked even without typing in the "developer on" command.

Here's what comes up with ioreg--

`skye@skyes-mbp ~ % ioreg | grep WUP-028 -A6
    | |   | | | +-o WUP-028@00100000  <class IOUSBHostDevice, id 0x1000060b0, registered, matched, active, busy 0 (62 ms), retain 31>
    | |   | | |   +-o AppleUSBHostCompositeDevice  <class AppleUSBHostCompositeDevice, id 0x1000060b5, !registered, !matched, active, busy 0, retain 4>
    | |   | | |   +-o IOUSBHostInterface@0  <class IOUSBHostInterface, id 0x1000060b6, registered, matched, active, busy 0 (49 ms), retain 12>
    | |   | | |   | +-o AppleUserUSBHostHIDDevice  <class AppleUserHIDDevice, id 0x1000060ba, registered, matched, active, busy 0 (8 ms), retain 13>
    | |   | | |   |   +-o IOHIDInterface  <class IOHIDInterface, id 0x1000060c0, registered, matched, active, busy 0 (6 ms), retain 6>
    | |   | | |   +-o Brave Browser  <class AppleUSBHostDeviceUserClient, id 0x1000060b7, !registered, !matched, active, busy 0, retain 7>
    | |   | | |   +-o Plex Tuner Servi  <class AppleUSBHostDeviceUserClient, id 0x1000060b8, !registered, !matched, active, busy 0, retain 6>
skye@skyes-mbp ~ % `

ryanmcgrath, 2022-04-26:
And to be clear, this is with the driver active?

Can you share the output of:

`sudo systemextensionctl list`

HomeStarRunnerTron, 2022-04-26:
Oh! No, that was with the driver uninstalled. \*reinstalls again* Here's the list with it on.

`    | |   | | | +-o WUP-028@00100000  <class IOUSBHostDevice, id 0x1000061b0, registered, matched, active, busy 0 (66 ms), retain 27>`
`    | |   | | |   +-o AppleUSBHostCompositeDevice  <class AppleUSBHostCompositeDevice, id 0x1000061b5, !registered, !matched, active, busy 0, retain 4>`
`    | |   | | |   +-o IOUSBHostInterface@0  <class IOUSBHostInterface, id 0x1000061b6, registered, matched, active, busy 0 (54 ms), retain 7>`
`    | |   | | |   | +-o GCAdapterDriverKitExtension  <class IOUserService, id 0x1000061ba, !registered, !matched, active, busy 0, retain 8>`
`    | |   | | |   +-o Brave Browser  <class AppleUSBHostDeviceUserClient, id 0x1000061b7, !registered, !matched, active, busy 0, retain 7>`
`    | |   | | |   +-o Plex Tuner Servi  <class AppleUSBHostDeviceUserClient, id 0x1000061b8, !registered, !matched, active, busy 0, retain 6>`
`    | |   | | +-o usb-drd0-port-ss@00200000  <class AppleT8103USB40XHCITypeCPort, id 0x1000001d0, registered, matched, active, busy 0 (18665 ms), retain 19>`

And here's systemextensionsctl list...

`skye@skyes-mbp ~ % sudo systemextensionsctl list`
`1 extension(s)`
`--- com.apple.system_extension.driver_extension
enabled	active	teamID	bundleID (version)	name	[state]`
`*	*	8JNDBG9U79	com.secretkeys.gcadapterdriverkitext (1.6/1)	com.secretkeys.gcadapterdriverkitext	[activated enabled]`

Continuing to get nothing in Dolphin, OpenEmu, etc.

ryanmcgrath, 2022-04-26:
Well, the driver is definitely attaching - `GCAdapterDriverKitExtension` is in the list of drivers there.

Which versions of Dolphin are you using? Some older versions (which some forks still are based on) lack hotplug support... I can't offhand recall which OpenEmu is using.

HomeStarRunnerTron, 2022-04-26:
I'm using the newest version of OpenEmu available from their GitHub, and I'm using Dolphin 5.0-15993. :  ) Should the controller come up in mappers like Enjoyable, etc.?

ryanmcgrath, 2022-04-26:
No, so the driver is simply about blocking Apple's drivers from attaching (which blocks programs from reading it). The adapter using the native protocol requires apps (e.g Dolphin) to have support for reading and using that native protocol - mappers won't necessarily do that though.

When you're looking in Dolphin, what's your adapter type set to in the Controllers config? It should be WiiU.

HomeStarRunnerTron, 2022-04-26:
WhoooOOPS! There we go, all I had to do was changed it to GameCube Adapter for Wii U, o, the foolishness of me.

[image]

Dolphin works perfectly, and if I plug in the other grey USB, I get rumble, even. Unbelievable. OpenEmu's Dolphin Core doesn't allow this, so-- I guess that Dolphin's the only program that allows it to be used at all, yes? ^_^ That's no small thing, obviously, (ESPECIALLY with how accurate it is), but maybe this should be addressed on your page (as well as in what situations it can be used, and how-- like to switch the Dolphin settings rather than try to map them when prompted), so foolish people such as myself don't think they might be able to use it anywhere else with a controller mapper and such? Thank you so, so much for your generosity.

closed 2022-04-26 by HomeStarRunnerTron

ryanmcgrath, 2022-04-26:
Don't quote me, but I want to say OpenEmu's Dolphin Core is older or something... so can't weigh in there.

It's worth noting that a Remapper _could_ read the adapter and do whatever with it (and close to a decade ago there was a driver that did this, actually). It's just that none do.

Anyway, glad it works! You may want to remove some of the images in this issue that have your name, by the way. Enjoy~