secretkeysio/GCAdapterDriver #9: Support for overclocking on macOS 11.0+

marcelroed, 2022-06-15:
Quoting the previous release in the "macOS 11.0+" section:
> This release is version 1.6; prior version numbers were lost in a battle with notarization and `systemextensionctl`. This release does not have an "overclock" ability yet, but should suffice for some users (overclock will come in an update).

What needs to be done to add overclocking in macOS 11.0+? I'm very interested in having this feature working, and can contribute if needed!

Thank you!

ryanmcgrath, 2022-06-15:
The feature's actually "done" in a local branch, but is ultimately stalled due to the API used for adjusting read rates not working properly in DriverKit. I reported it back in Big Sur but don't believe it's been fixed yet - though here's hoping Ventura does it.

At this point I should _probably_ just sign the overclocked kext for 11.0+ because I don't see how to do this otherwise.

If you want to overclock on Big Sur/Monterey/Ventura, and are comfortable disabling SIP, there _is_ a guide on doing it the old school way over at the project-slippi Wiki: https://github.com/project-slippi/Ishiiruka/wiki/Overclocking-Controllers-on-macOS-Big-Sur-or-Monterey

Note that disabling SIP isn't really great since it'll compromise certain DRM-based feed quality, and thinks like Apple Pay will have reduced or no functionality on some Mac models... which is why I should probably just sign the kext and be done with it.

marcelroed, 2022-06-19:
Interesting, thank you for your response! I've disabled SIP for other reasons before and ended up reverting due to having some issues with some functionality, as you mentioned.

If you could sign the overclocked kext, that would be awesome.

Thanks again!

ryanmcgrath, 2022-06-19:
If I don't get to it this week, feel free to ping me here again - I've been crazy busy unfortunately so I'll have to fit it in somewhere is all.

marcelroed, 2022-06-19:
Great! Thanks for your good work.

alanonthegit, 2022-08-02:
Are there any updates on this?

ryanmcgrath, 2022-08-02:
No, I'm in an annoyingly busy period of weeks right now. I realistically can't look at this until Black Hat & DefCon are over. :(

marcelroed, 2022-10-30:
Hey @ryanmcgrath ! Any chance of getting this done? I'm on Ventura now; not sure if they fixed this bug or if you're able to sign the kext.

ryanmcgrath, 2022-11-09:
Haven't forgotten about this, just been too busy with work + trying to fix Slippi on Ventura. I need to confirm if this was actually fixed on Ventura period...

gganley, 2022-12-15:
https://github.com/secretkeysio/GCAdapterDriver/issues/9#issuecomment-1156973587

mind opening this in a PR Slippi was just updated to work on Ventura

ryanmcgrath, 2023-06-19:
I haven't, no - both this and some other open source work I've been trying to get time for have unfortunately fallen by the wayside due to my day to day work. :(

Are you an M1 or Intel device?

MasterRee, 2023-06-19:
Thanks for the quick reply. The work that puts food on the table comes first, so I understand.

> Are you an M1 or Intel device?

I'm on an M1 Pro running macOS 13.4.

ZXSwire3, 2023-09-17:
I'm on an M2 Max running macOS 13.5.2. 
I'm not sure if there is anything that I could do to help out, but I would be more than willing to if there is anything :)

ZigAUM, 2024-04-07:
Also hoping for an update on this- on an M2 running the latest OS, would love to be able to overclock my Wii U adapter. . . What exactly needs to be done to enable this functionality? 

Thank you.

ryanmcgrath, 2024-04-07:
You still _can_, GitHub just arbitrarily changes wiki links: https://github.com/project-slippi/Ishiiruka/wiki/Overclocking-Controllers-on-macOS-Big-Sur,-Monterey,-Ventura,-or-Sonoma

This particular feature requires me doing some work that I just don't have time for. It's not something that someone can contribute as it primarily deals with signing the driver itself.