secretkeysio/GCAdapterDriver #10: Error: Error Domain=OSSystemExtensionErrorDomain Code=10 "(null)"

tejomaygadgil, 2022-06-24:
Getting this error when I toggle GCAdapter on: 

```
Failed to activate extension!

Error: Error Domain=OSSystemExtensionErrorDomain Code=10 "(null)"

Details: The operation couldn’t be completed. (OSSystemExtensionErrorDomain error 10.)
```

I'm running it on a MacBook Pro (13-inch, M1, 2020) running MacOS Monterey.

[image]

ryanmcgrath, 2022-06-24:
`Code 10` indicates your system isn't allowed to install extensions like this. Are you on a managed device (work, school, parents)...?

tejomaygadgil, 2022-06-24:
Yeah ;(

Any workarounds?

ryanmcgrath, 2022-06-24:
Unfortunately no - the driver can't really work around that restriction. :(

If you're looking to play on Dolphin and your adapter supports "PC Mode" (via a hardware switch), you can set it up under "Standard Controller" and use it that way since it doesn't require a driver. Not all adapters support this though (Mayflash, for instance, does).

closed 2022-06-24 by ryanmcgrath