secretkeysio/GCAdapterDriver #12: "wiiu-gcc-adapter-bin" wants to make changes. Enter password to allow this. + Error Opening Adapter: Access denied (insufficient permissions)

cyndavilachase, 2022-11-30:
Getting this error message as previously described in [this other closed issue post](https://github.com/secretkeysio/GCAdapterDriver/issues/3): 

[image]

The error seems to come from this, which keeps popping up even if you input the password:

[image]

However, any permissions I mess with don't seem to solve the issue. Also tried renewing keychain access to it's default and tried disabling SIP to no avail. Plugged and unplugged the adapter many times and the pop-up just won't go away no matter what I try.

Any help would be appreciated. Tried posting a thread to reddit but haven't had any luck yet, so thought I'd try for the source of this driver.

ryanmcgrath, 2022-11-30:
Mmm, `wiiu-gcc-adapter-bin` isn't anything from this repository. Where'd you get that file?

ryanmcgrath, 2022-11-30:
Oh I see the tab in the background - that's an older driver that can't be installed alongside GCAdapterDriver. You'll want to remove/uninstall that so GCAdapterDriver is the only one.

See: https://secretkeys.io/gcadapterdriver/ in the section `How do I remove osx-wiiu-gcc-adapter?`

ryanmcgrath, 2022-11-30:
I assume this is Catalina or Mojave, based on the window UI? Or is it Big Sur or Monterey...?

cyndavilachase, 2022-11-30:
Monterey.

cyndavilachase, 2022-11-30:
Just tried that. When I click on the app icon nothing happens. Thought that meant it was just running and I didn't have to do anything else.

ryanmcgrath, 2022-11-30:
Yup, it runs an app in the menubar of your Mac that handles loading the driver. In the future it won't be needed, theoretically, but just watch for the icon up there. You probably need to also unplug + replug in your adapter (the USB side, not the controller side) to get it to register properly.

cyndavilachase, 2022-11-30:
Done. It's working now! Thank you so much.

ryanmcgrath, 2022-11-30:
Congrats!

closed 2022-11-30 by ryanmcgrath