secretkeysio/GCAdapterDriver #13: Official WiiU adapter MacOS Ventura

bloodstrike, 2022-12-07:
Is there any issues or tricks with getting an official adapter to work these days? Using the .app version, I've activated the driver. Using a mayflash adapter, I can see the controller and use it in different applications with no issues. However, if I instead use the official WiiU branded adapter, controllers never show up. Tried different usb ports, restarting the computer, etc with no changes. The adapter shows up in system information as "WUP-028" under USB devices.

ryanmcgrath, 2022-12-07:
What macOS version + hardware combo?

ryanmcgrath, 2022-12-08:
Hmm, that should work fine. To be clear, you're activating the driver by hitting the menu bar icon for the dropdown, then hitting the toggle?

And then unplugging the _adapter_ and plugging it back in?

(Bear with the day 1 questions, hard to debug remotely lol)

bloodstrike, 2022-12-09:
Apologies for the delay, and apologies for more 😉

I was testing more and trying to figure out what I was missing, and it turns out what I was missing was basic reading comprehension. I had read a few things, and was under the impression that Dolphin had a *built in* driver for the GC adapter, and that the other drivers (this one, smash enabler and the like) were only needed to get the adapter working with *other* applications. Trying to figure out what was wrong with my setup, I actually read through all the information that I could find on Dolphin forums and other places, and realized that no, these drivers were needed specifically so that Dolphin's implementation would work, and *only* dolphin due to the whole HID driver situation stuff with MacOS. So with that knowledge and knowing what to expect, everything is working correctly as expected. With the driver installed and enabled, the Dolphin 'GameCube Adapter for WiiU' option works correct with no issues.

Again, apologies for my misunderstanding. Thanks for making this easy to install and use driver!

closed 2022-12-09 by bloodstrike

ryanmcgrath, 2022-12-09:
No worries - glad you're up and running!