secretkeysio/GCAdapterDriver #17: Expose GC Controllers to system as Gamepads?

HenningSchaar, 2023-02-11:
Hi! 
First of all thank you for making your software super easy to use, I was just wondering if you know of any ways to get macos to recognise the WUP-028 as virtual Gamepads (like the old project by hjelmn: gcusbadapter)? Any leads welcome, since I would really love to use my GC Controllers in other applications than Dolphin (although I understand that is not the goal of this software).
(using m1 pro machine with ventura)

seanmcnally98, 2024-07-18:
I'll second this.  I actually downloaded this software thinking this is what it does.  Looks like that [hjelmn project](https://github.com/hjelmn/gcusbadapter_osx/releases) you mentioned hasn't been updated since 2015, but then again, I bought my mac that year...I'll try it and see what happens.

ryanmcgrath, 2024-07-18:
I don't know of any particular way to do that and I'm not really inclined to support it, since it's an ultra-niche use case.

Someone is free to PR it, but with DriverKit being the way it is I feel like it'd be tricky to develop outside of this org - but perhaps with a properly disabled SIP + systemextensionctl setup it's possible.

You might be able to get away with using just `gcusbadapter.kext` from the hjelmn project (with SIP kext checks disabled only, don't disable all of SIP necessarily), since GCAdapterDriver handles the part that `gcusbadapter_raw.kext` does. kexts are not guaranteed to work forever though, so... YMMV.

jneen, 2025-04-18:
Hi there!

Just adding my hat to the ring to say it may not be quite so ultra-niche - I for instance would love to be able to use my gamecube controller for Celeste and other Steam games.