secretkeysio/GCAdapterDriver #18: Can't find the app to uninstall

Wado518, 2023-02-20:
I would like to uninstall GCAdapterDriver. The FAQ under How do I Uninstall? says 

> Move the app to the Trash, then clear your Trash. You can optionally disable the driver from within the app if you want to just stop the detection without removing the app itself.

Where is the app located on my computer? I checked the applications folder and even tried a global search and can't find anything called GCAdapterDriver.app.

in ~/Library/Extensions, there is a GCAdapterDriverPassThrough.kext file. Do I have to do anything with that?

(I'm on Mac Catalina)

Thanks 🙂

ryanmcgrath, 2023-02-20:
MacOS Catalina is `10.x`, so follow the instructions for that, not the `11.0+` instructions. :)

>A bit trickier! You can copy and paste the following commands into a Terminal.app instance.
>
>sudo kextunload /Library/Extensions/GCAdapterDrive*.kext
>sudo rm -rf /Library/Extensions/GCAdapterDrive*.kext
>
>You may also want to reboot after this.

Note that the driver does 100% nothing without a device plugged in, so you don't necessarily need to uninstall it unless you're trying to replace it with the overclocked one or something. Whatever the case, though, I hope the above is helpful!

ryanmcgrath, 2023-02-20:
Also, `copy and paste` is highly recommended - I hate that there's no easy way to avoid having users run a `rm -rf`...

closed 2023-02-20 by ryanmcgrath

Wado518, 2023-02-23:
Thanks! Basically I saw that my macos version was 10.15 and I only looked at the 15 so I thought I needed to follow the 11.0+ instructions haha.