secretkeysio/GCAdapterDriver #19: Installation Process

JJMississippi1, 2023-05-02:
When I installed the GCAdapter Driver Installer, I dragged the icon to the application as it said in the instructions but when I opened it, it was the installer and not the driver. I can only get the installer but never the driver.

ryanmcgrath, 2023-05-02:
Can you provide screenshots of what you're seeing? That doesn't sound... possible.

JJMississippi1, 2023-05-02:
￼
I would drag the app into my applications and then it opens to this

JJMississippi1, 2023-05-02:
I got the Driver activated but Dolphin still doesn't detect the adapter. Should I take the adapter out and plug it in again or should I reboot Dolphin?

ryanmcgrath, 2023-05-02:
Dolphin or Slippi Dolphin?

And yes, you have to unplug and replug - the guide notes this.

JJMississippi1, 2023-05-02:
Dolphin, I did that and it still didn't work. I restarted my Mac and did the same thing but nothing different.

ryanmcgrath, 2023-05-02:
- What adapter are you using?
- If it's a Mayflash, is it in WiiU mode (there's a switch on the side of the adapter)?
- When you activated the driver, was it just via running the app, or did you also flip the toggle in the menu icon dropdown?

JJMississippi1, 2023-05-02:
The adapter is from Syntech and there are no switches on the adapter. I flipped the toggle on the dropdown menu and it said it was activated after letting it be accepted by system preferences.

ryanmcgrath, 2023-05-02:
Hmm, I don't know if anybody has ever tried a Syntech with this driver - frankly I've never heard of the brand.

Can you check what the hardware is identifying it as by going to:

- Apple Menu
- About This Mac
- More Info
- System Report
- USB

JJMississippi1, 2023-05-03:
￼
Sorry it took so long, I was doing finals for college. Here’s a screenshot of what you told me to get to.

ryanmcgrath, 2023-05-03:
If you're attaching it via Email, GitHub won't accept it - needs to be attached via the Web UI.

JJMississippi1, 2023-05-03:
[image]
Does this work?

ryanmcgrath, 2023-05-03:
Yes, though - can you click on the gamepad rather than the USB line?

JJMississippi1, 2023-05-04:
[image]
This?

JJMississippi1, 2023-05-04:
[image]
Wrong Screenshot, sorry

ryanmcgrath, 2023-05-04:
So that unfortunately looks like it uses an odd vendor/product ID combo, which is why the driver doesn't work.

Have you tried configuring it as a `Standard Controller` in Dolphin? That's the other way to go about this, and is fine for 99% of whatever you'd be doing - it's really mostly Melee players that might care about native response times.

JJMississippi1, 2023-05-04:
Ok it worked, thank you

closed 2023-05-04 by JJMississippi1