secretkeysio/GCAdapterDriver #21: Cemu for macOS

GreetingsIExist, 2023-12-31:
As far as I know, this work is more focused on getting the adapters working on Dolphin.

Is it possible to get this driver to work on the Wii U emulator Cemu for macOS? More specifically for the Super Smash Bros. for Wii U game, which allows the usage of GameCube controllers through an adapter?

Have asked regarding support on the Cemu Discord server and the reply I received from a mod was, quote "No idea! I don't think anyone has tried." "If they show up in the Mac's Settings/Controllers panel, you should be good to go with SDLcontroller at the Cemu side."

ryanmcgrath, 2023-12-31:
This driver mostly stops macOS from grabbing the device and making apps unable to read it; CEMU would need to add support for natively reading it - or if your adapter supports “PC mode”, SDL should be able to pick that up.

GreetingsIExist, 2023-12-31:
The adapter that I'm trying to use is the original one, so no PC/Wii U switch. With the driver, the adapter only "turns on" when using Dolphin. Otherwise it's just connected and detected on macOS but not usable. Even with the adapter being "activated" with Dolphin, it cannot be used outside of Dolphin.

So, is that the issue? Using the official adapter with no PC/Wii U switch, therefore not being usable with SDL?

GreetingsIExist, 2023-12-31:
That being said, you're right. There's no "native support" like the one Dolphin has.
Presented a "request" regarding "native support" on Cemu's Discord server. Will see depending on their response if the idea may be presented on the "Issues" section of their GitHub.

GreetingsIExist, 2023-12-31:
Since this request is more on Cemu's side, I shall close this issue.

closed 2023-12-31 by GreetingsIExist