secretkeysio/GCAdapterDriver #23: GCAdapterDriver not wokring over the apple one

Norcyy, 2024-04-04:
I've followed all of the instructions, i can see that both Dolphin and the computer registers my adapter, but the driver from the app doesn't seem to connect in favour of the apple one. I've done the unplug and replug, I've deleted and redownloaded, and I'm just lost on what to do, and i can't seem to find any other information on troubleshooting besides the adapter not picking up.

[image]

ryanmcgrath, 2024-04-04:
Ah, to be clear you're trying to install the `.kext`? Need to know your actual macOS version before we could even begin to debug - based on Finder I'm guessing it's 10.15 or less?

Norcyy, 2024-04-04:
I assume that may be what i need? i’m on macOS Monterey (version 12.7.2) on a 2019 macbook air

ryanmcgrath, 2024-04-04:
Monterey is not a `.kext`-based installation method - the GCAdapterDriver homepage indicates this:

[image]

You're trying to run the `.kext` installer from the right, but you _actually_ need to do the installer on the left.

Norcyy, 2024-04-04:
Oh yeah, I had actually used the left one. My .kext assumption was based on other comments people had left. I opened the installer, dragged it to applications, opened from my applications folder in finder. I went to system privacy to allow it to run and in my top bar it said it was activated. I was able to follow another issue up until about the halfway point when the command they ran that brought up the results in my first screenshot worked, and it showed the correct driver. Mine however, still shows the apple driver no matter how many times i unplug and replug. I should also thank you a ton for the quick response! something i definitely wasn’t expecting

ryanmcgrath, 2024-04-04:
What's the output of `sudo systemextensionsctl list` when the driver is toggled on from the dropdown?

Norcyy, 2024-04-04:
[image]

I'm currently unable to test the gc adapter itself until a few hours, this is what my list looks like. I assume it's duplicated from the times I've turned it off and on again

ryanmcgrath, 2024-04-04:
I'd go ahead and do a full reboot to clear out any of those older installs - they _shouldn't_ be conflicting, but frankly DriverKit is weird like this and I've run into cases where people have it "just work" after a full reboot.

(DriverKit doesn't fully remove a driver install until a reboot occurs)

Norcyy, 2024-04-04:
might be a silly question, but i assume a full reboot is just a restart? I've done this before and it didn't work, but things may have been different which i don't remember so I'll disable it and try it again

ryanmcgrath, 2024-04-04:
Also - what adapter type are you attempting to connect?

Norcyy, 2024-04-04:
I'm using an official switch gamecube adapter (which i assume is the same one as the wii U)? Anyway, i restarted and now the command runs this -

[image]

Norcyy, 2024-04-04:
haha, yep, it's probably important to check those. I have definitely been using the black cord the entire time, and I'm using the standard apple usb-c to usb-a adapter

[image]

The adapter itself seems to connect fine, it shows up as WUP-028 in the system report, and when checking dolphin it says that there is an adapter detected. It's just that inputs won't register

ryanmcgrath, 2024-04-04:
Screenshot your controller(s) panel from Dolphin~

Norcyy, 2024-04-04:
ah, i currently don't have access to my controller / adapter or anything so I'm not sure anything interesting will show up but I'll post anyway and update in a few hours when i can try with the adapter connected.

[image]

Norcyy, 2024-04-05:
Alright, here we go. It says that the adapter is connected, but the controller will not register anything in emulation, specifically when i try to press the start button to enter the game

[image]

ryanmcgrath, 2024-04-05:
This is plugged into port 1, correct? If it's detect, it should be outputting a Hz measurement...

Norcyy, 2024-04-05:
is port 1 the furthest one to the back (the left one when looking straight on)? I've been using the other one, but trying it with that one doesn't seem to work either. In between testing these things do I need to reset each time? and where may the Hz measurement show up if it does work?

ryanmcgrath, 2024-04-05:
Just set all your ports in that config panel to WiiU adapter, then each port would technically work.

Right now you only have one port enabled in the config, and the adapter says detected but not reading... which is why I ask this.

Norcyy, 2024-04-05:
Ohh i'm understanding it now, I thought you were talking about the computer ports.

and OMG it works!!! It turns out that I had thought port 1 was the one that follows directly through the adapter cable, which is actually the last one. Thanks so much for helping, though these computer related issues do always seem to be caused by the little things that are always glossed right over. I wish you the very best!

closed 2024-04-05 by Norcyy