secretkeysio/GCAdapterDriver #25: Troubleshooting: Adapter Not Detected Sonoma

gccontrolr, 2024-05-21:
I installed the [GCAdapterDriver ](https://secretkeys.io/gcadapterdriver/) macOS 11.0+ (v1.6). I am running Sonoma on a 2023 macbook pro. So far, I have been unable to detect the gc controller in Dolphin (5.0), only my default mouse pad appears when in the "configure" setting. The hardware I am using are 1) game cube controller ->  an official gc controller to usb adapter -> USB to USB-C converter. Can someone please help me troubleshoot? 

[image]
[image]

ryanmcgrath, 2024-05-21:
You're in the settings screen for the standard controller. You need to select "WiiU Controller" or whatever it is so that you get the native GC Controller protocol.

gccontrolr, 2024-05-21:
oh my, that was it... Thank you!

closed 2024-05-21 by gccontrolr