secretkeysio/GCAdapterDriver #26: Adding support for Skylanders Portals of Power

jamescoats, 2024-07-21:
Would love to see support for the physical Skylanders Portals of Power to work under Dolphin on Apple Silicon. It seems like this may be the best way to get it going. An initial request was started [here](https://github.com/secretkeysio/GCAdapterDriver/issues/5), but thought it would be better in its own request.

I'd love to help with this - I own a Trap Team Portal of Power and will do whatever I can.

jamescoats, 2024-07-31:
Hopefully these details can help

Traptanium Portal Details:
FCC ID: XLU0000586 
IC: 7293B-0000586
Model No. 0000586

These are the details from System Information from macOS 14.4.1 on my Apple M1 Pro:

USB 3.1 Bus:
Host Controller Driver:	AppleT6000USBXHCI
Spyro Porta:

  Product ID:	0x0150
  Vendor ID:	0x1430  (RedOctane)
  Version:	1.00
  Speed:	Up to 12 Mb/s
  Manufacturer:	Activision
  Location ID:	0x02100000 / 1
  Current Available (mA):	500
  Current Required (mA):	500
  Extra Operating Current (mA):	0

Let me know if I can help with more information or testing etc.

sanjay900, 2025-01-31:
This was just built, see if it works https://secretkeys.io/gcadapterdriver/GCAdapterDriver-1.8-beta.app.zip

SpeedForceGOD, 2025-02-01:
Hey so im abit new, i got dolphin and my portal, ia slo downloaded the gcadapter, what do I need to do to play swap force? when I try to load it my portal keeps saying its connected and then disconnected can you please help me out?

> This was just built, see if it works https://secretkeys.io/gcadapterdriver/GCAdapterDriver-1.8-beta.app.zip

ryanmcgrath, 2025-02-01:
When I last tried getting these devices to work, they required additional power setting keys that this build doesn’t have. I’ll find time to add them to another test build in the future - but watch the other thread for updates.

SpeedForceGOD, 2025-02-01:
> When I last tried getting these devices to work, they required additional power setting keys that this build doesn’t have. I’ll find time to add them to another test build in the future - but watch the other thread for updates.
> […](#)

Sounds good thank you so much also which thread are you referring to sorry its my first time on here

ryanmcgrath, 2025-02-01:
The one with the 1.8-beta build linked. :)

ryanmcgrath, 2025-02-01:
Sorry, didn't realize it wasn't linked. I meant this thread: https://github.com/secretkeysio/GCAdapterDriver/pull/27

jamescoats, 2025-02-02:
> This was just built, see if it works https://secretkeys.io/gcadapterdriver/GCAdapterDriver-1.8-beta.app.zip

Amazing, thanks for the effort @sanjay900 and @ryanmcgrath! 

Can't get the portal going as yet, possibly to do with the power settings issue outlined [here](https://github.com/secretkeysio/GCAdapterDriver/issues/26#issuecomment-2629116842)

Here are the details for me to reproduce, please let me know if I'm missing something:

macOS 15.3
Dolphin Release 2412
Skylanders: Spyro's Adventure (SSPE52) ROM
Traptanium Portal (see details [here](https://github.com/secretkeysio/GCAdapterDriver/issues/26#issuecomment-2259793246))

1. Downloaded the 1.8-beta build. Ran it and toggled it on. Toggled on driver extension in System Settings
2. Plugged in Portal of Power
3. Launched Dolphin
4. In Dolphin settings, under "Wii" tab, added "1430:0150 - Skylanders Portal" to the "Whitelisted USB Passthrough Devices"
5. Started up Spyro's Adventure
6. Got to the point where it says "Press A to begin".
7. ROM displays white screen and I'm unable to continue

This white screen only appears when #4 is active, otherwise it says to plug the portal in

jamescoats, 2025-02-02:
Sorry, pretty new to this part of GitHub and keep on accidentally closing issue.

SpeedForceGOD, 2025-02-02:
> > This was just built, see if it works https://secretkeys.io/gcadapterdriver/GCAdapterDriver-1.8-beta.app.zip
> 
> Amazing, thanks for the effort [@sanjay900](https://github.com/sanjay900) and [@ryanmcgrath](https://github.com/ryanmcgrath)!
> 
> Can't get the portal going as yet, possibly to do with the power settings issue outlined [here](https://github.com/secretkeysio/GCAdapterDriver/issues/26#issuecomment-2629116842)
> 
> Here are the details for me to reproduce, please let me know if I'm missing something:
> 
> macOS 15.3 Dolphin Release 2412 Skylanders: Spyro's Adventure (SSPE52) ROM Traptanium Portal: Product ID: 0x0150, Vendor ID: 0x1430 (RedOctane). See more details [here](https://github.com/secretkeysio/GCAdapterDriver/issues/26#issuecomment-2259793246))
> 
> 1. Downloaded the 1.8-beta build. Ran it and toggled it on. Toggled on driver extension in System Settings
> 2. Plugged in Portal of Power
> 3. Launched Dolphin
> 4. In Dolphin settings, under "Wii" tab, added "1430:0150 - Skylanders Portal" to the "Whitelisted USB Passthrough Devices"
> 5. Started up Spyro's Adventure
> 6. Got to the point where it says "Press A to begin".
> 7. ROM displays white screen and I'm unable to continue
> 
> This white screen only appears when number 4 is active, otherwise it says to plug the portal in

Hey just to provide some more information to see if this helps from my end as well:

Product ID:	0x0150
  Vendor ID:	0x1430  (RedOctane)
  Version:	25.08
  Speed:	Up to 12 Mb/s
  Manufacturer:	Activision
  Location ID:	0x00120000 / 5
  Current Available (mA):	500
  Current Required (mA):	300
  Extra Operating Current (mA):	0

I'm playing Skylanders Swap Force Wii edition on my Mac, on the portal it says that the required amount of current is provided in the picture below. When I tried using the new beta for GCAdapter and then whitelisted the USB device, the game started to loop and the portal was disconnected and connected which is also sent below. I tried to start the game and hope it will clear up when I'm in but it does the same thing once I'm in the game as well.

[image]

https://github.com/user-attachments/assets/595ac99f-0e17-484b-a6a4-fbd14209bdda

sanjay900, 2025-02-02:
Would i be able to get a USB descriptor dump from one of you guys from the portal?
Tempted to emulate one on a micro controller and see if that has the power issues too or not

SpeedForceGOD, 2025-02-02:
> Would i be able to get a USB descriptor dump from one of you guys from the portal? Tempted to emulate one on a micro controller and see if that has the power issues too or not
> 
> An HID descriptor dump would also be interesting too.

I don't mind doing anything to help, if you could just briefly explain how I can retrieve that I'll send it over!

sanjay900, 2025-02-02:
Have you got a windows box handy or just the mac?

SpeedForceGOD, 2025-02-02:
> Have you got a windows box handy or just the mac?

I have a windows VM on my mac and a windows laptop with me aswell

sanjay900, 2025-02-02:
For the hid descriptor, this should work on the mac: https://github.com/todbot/mac-hid-dump

For the usb descriptor, i tend to get people to use https://www.thesycon.de/eng/usb_descriptordumper.shtml but thats windows only, not sure what to use for macos.

sanjay900, 2025-02-02:
Cool, if you could use those two apps to dump the hid and usb descriptors that would be great.

ryanmcgrath, 2025-02-02:
If nothing else, I have a local branch on my old Intel machine that has the values. One person successfully tested it (kext) in the past and confirmed it's possible for it to work, I'd just need to find time to grab the values.

sanjay900, 2025-02-02:
What exactly is the power issue in question out of curiousity?

ryanmcgrath, 2025-02-02:
Those devices appear to need some values for USB power control, otherwise they go through a constant loop of connecting-disconnecting. I don't own one myself so I only know this from debugging with another user some years ago (their dump of values is what I wound up matching to).

SpeedForceGOD, 2025-02-02:
> Cool, if you could use those two apps to dump the hid and usb descriptors that would be great.

here is the hid:

1430 0150: Activision - Spyro Portal
DESCRIPTOR:
  06  00  ff  09  01  a1  01  19  01  29  40  15  00  26  ff  00 
  75  08  95  20  81  00  19  01  29  40  91  00  c0  
  (29 bytes)

as for the USB one I will have to give to you in a bit since I need to find my windows laptop

sanjay900, 2025-02-02:
How odd, it seems that app is only outputting partial hid descriptors :(

May just get you to run https://github.com/todbot/win-hid-dump on the windows laptop then, it should give me the whole report

SpeedForceGOD, 2025-02-02:
> How odd, it seems that app is only outputting partial hid descriptors :(
> 
> May just get you to run https://github.com/todbot/win-hid-dump on the windows laptop then, it should give me the whole report
Whoops that might be my bad I only selected the output for the portal not for my other peripherals, are you referring to that?

sanjay900, 2025-02-02:
Nah, the hid descriptor usually would contain a lot more than that but that dumping tool has only extracted part of it

SpeedForceGOD, 2025-02-02:
> Nah, the hid descriptor usually would contain a lot more than that but that dumping tool has only extracted part of it

No worries ill run it on my windows ill be back around 15 mins

SpeedForceGOD, 2025-02-02:
> Nah, the hid descriptor usually would contain a lot more than that but that dumping tool has only extracted part of it

The HID tester opens a terminal shows the descipt briefly then closes quickly i cant seem to copy anything

SpeedForceGOD, 2025-02-02:
> Nah

here is the usb descript: Information for device Spyro Portal (VID=0x1430 PID=0x0150):

------------------------------
Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device supports USB 2.0 specification
Device address: 0x0005
Current configuration value: 0x01
Number of open pipes: 2

------------------------------
Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0200	bcdUSB
0x00	bDeviceClass      
0x00	bDeviceSubClass   
0x00	bDeviceProtocol   
0x20	bMaxPacketSize0   (32 bytes)
0x1430	idVendor
0x0150	idProduct
0x2508	bcdDevice
0x01	iManufacturer   "Activision"
0x02	iProduct        "Spyro Portal"
0x00	iSerialNumber
0x01	bNumConfigurations

-------------------------
Configuration Descriptor:
-------------------------
0x09	bLength
0x02	bDescriptorType
0x0029	wTotalLength   (41 bytes)
0x01	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0x80	bmAttributes   (Bus-powered Device)
0x96	bMaxPower      (300 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0x03	bInterfaceClass      (Human Interface Device Class)
0x00	bInterfaceSubClass   
0x00	bInterfaceProtocol   
0x00	iInterface

HID Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
0x0111	bcdHID
0x00	bCountryCode
0x01	bNumDescriptors
0x22	bDescriptorType   (Report descriptor)
0x001D	bDescriptorLength

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress  (IN endpoint 1)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0020	wMaxPacketSize    (1 x 32 bytes)
0x04	bInterval         (4 frames)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x01	bEndpointAddress  (OUT endpoint 1)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0020	wMaxPacketSize    (1 x 32 bytes)
0x04	bInterval         (4 frames)

Microsoft OS Descriptor is not available. Error code: 0x0000001F

--------------------------------
String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "Activision"
0x02   0x0409  "Spyro Portal"

------------------------------

Connection path for device: 
USB xHCI Compliant Host Controller
Root Hub
Spyro Portal (VID=0x1430 PID=0x0150) Port: 1

Running on: Windows 10 or greater (Build Version 22631)

Brought to you by TDD v2.19.0, Dec  5 2023, 12:08:38

sanjay900, 2025-02-02:
> > Nah, the hid descriptor usually would contain a lot more than that but that dumping tool has only extracted part of it
> 
> The HID tester opens a terminal shows the descipt briefly then closes quickly i cant seem to copy anything

Open a command prompt first and then run it that way

SpeedForceGOD, 2025-02-02:
> > > Nah, the hid descriptor usually would contain a lot more than that but that dumping tool has only extracted part of it
> > 
> > 
> > The HID tester opens a terminal shows the descipt briefly then closes quickly i cant seem to copy anything
> 
> Open a command prompt first and then run it that way

doesnt seem to work its still doing the same

sanjay900, 2025-02-03:
One more question
what happens if you grab a copy of https://github.com/todbot/hidapitester/releases/tag/v0.5
and run `hidapitester --vidpid 1430:0150 --open --read-feature 0 `

SpeedForceGOD, 2025-02-03:
> One more question what happens if you grab a copy of https://github.com/todbot/hidapitester/releases/tag/v0.5 and run `hidapitester --vidpid 1430:0150 --open --read-feature 0 `

could I use this same cmd on mac?

sanjay900, 2025-02-03:
Yeah, should work just fine on macos

jamescoats, 2025-02-03:
Hi, here is the output for the HID descriptors and USB descriptors for the Traptanium Portal. Please note that I used Parallels Desktop to run the Windows programs.

## HID Descriptor

`./hidapitester --vidpid 1430:0150 --open --read-feature 0`

```
1430 0150: Activision - Spyro Porta
DESCRIPTOR:
  06  00  ff  09  01  a1  01  19  01  29  40  15  00  26  ff  00 
  75  08  95  20  81  00  19  01  29  ff  91  00  c0  
  (29 bytes)
```

`./hidapitester --vidpid 1430:0150 --open --get-report-descriptor`

```
Opening device, vid/pid: 0x1430/0x0150
Report Descriptor:
 06 00 FF 09 01 A1 01 19 01 29 40 15 00 26 FF 00 75 08 95 20 81 00 19 01 29 FF 91 00 C0
Closing device
```

`winhiddump-0.3` (Using Parallels Desktop). This has more data than the above HID descriptor

```
1430:0150: Activision - Spyro Porta
PATH:\\?\hid#vid_1430&pid_0150#7&2b90a745&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  06  00  FF  09  01  A1  01  19  01  29  40  15  00  25  01  35
  00  45  01  65  00  55  00  75  01  95  40  81  00  95  C0  81
  03  19  01  29  FF  95  FF  91  00  95  01  91  03  C1  00
  (47 bytes)
```

## USB Descriptor

Thesycon USB Descriptor Dumper (Using Parallels Desktop)

```
Information for device Spyro Porta (VID=0x1430 PID=0x0150):

------------------------------
Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device address: 0x0006
Current configuration value: 0x01
Number of open pipes: 2

------------------------------
Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0200	bcdUSB
0x00	bDeviceClass      
0x00	bDeviceSubClass   
0x00	bDeviceProtocol   
0x40	bMaxPacketSize0   (64 bytes)
0x1430	idVendor
0x0150	idProduct
0x0100	bcdDevice
0x01	iManufacturer   "Activision"
0x02	iProduct        "Spyro Porta"
0x00	iSerialNumber
0x01	bNumConfigurations

-------------------------
Configuration Descriptor:
-------------------------
0x09	bLength
0x02	bDescriptorType
0x0029	wTotalLength   (41 bytes)
0x01	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0x80	bmAttributes   (Bus-powered Device)
0xFA	bMaxPower      (500 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0x03	bInterfaceClass      (Human Interface Device Class)
0x00	bInterfaceSubClass   
0x00	bInterfaceProtocol   
0x00	iInterface

HID Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
0x0111	bcdHID
0x00	bCountryCode
0x01	bNumDescriptors
0x22	bDescriptorType   (Report descriptor)
0x001D	bDescriptorLength

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress  (IN endpoint 1)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x01	bInterval         (1 frames)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x02	bEndpointAddress  (OUT endpoint 2)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x01	bInterval         (1 frames)

Microsoft OS Descriptor is not available. Error code: 0x0000001F

--------------------------------
String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "Activision"
0x02   0x0409  "Spyro Porta"

------------------------------

Connection path for device: 
USB xHCI Compliant Host Controller
Root Hub
Generic USB Hub
Spyro Porta (VID=0x1430 PID=0x0150) Port: 13

Running on: Windows 10 or greater (Build Version 22621)

Brought to you by TDD v2.19.0, Dec  5 2023, 12:08:38

```

sanjay900, 2025-02-03:
I suppose it makes a lot of sense that the portals just don't follow any of the standards for a normal PS controller

jamescoats, 2025-02-04:
> Even the xbox 360 portals are totally different to a normal controller, which is interesting because i didnt even know the xbox 360 specs let you do that lol

Oh right, I also have an Xbox One specific Traptanium Portal, but I assumed getting it working was off the table. I can send through the details tonight.

sanjay900, 2025-02-04:
I have all the details for the xb1 and 360 portals funnily enough.
If we wanted to make the other portals work there are options for that, but they don't need anything done here.
I myself have done that for making the PS3 and xbox 360 instruments work on the wii, so if i was gonna do it, id probably do something similar to the logic i used there.

jamescoats, 2025-02-04:
Pretty sure the PS3/PS4 and Wii/Wii U portals are the same. Just the 360 and Xbox One are different and incompatible with Wii/PS. Correct me if I'm wrong though

sanjay900, 2025-02-04:
Correct, we would only have to write a module for the xbox portals if we wished to make them work on wii

jamescoats, 2025-04-12:
@sanjay900 Any chance we can get a release build of your fork to test?

sanjay900, 2025-04-12:
> [@sanjay900](https://github.com/sanjay900) Any chance we can get a release build of your fork to test?

https://secretkeys.io/gcadapterdriver/GCAdapterDriver-1.8-beta.app.zip
This is the build for testing right now

jamescoats, 2025-04-17:
> https://secretkeys.io/gcadapterdriver/GCAdapterDriver-1.8-beta.app.zip This is the build for testing right now

Ah thanks - tested it again and got the same results as the first time here: https://github.com/secretkeysio/GCAdapterDriver/issues/26#issuecomment-2629298147

sanjay900, 2025-04-17:
Yeah the change I made recently wasn't for portals, it was for a different device

jamescoats, 2025-12-14:
@deReeperJosh any way we can test this build? Which portal did you test? Is there a way we can sign ourselves via command-line? Cheers

sanjay900, 2025-12-18:
what ids did you actually change in your build? the diff is not fun to read haha

or was that simply just reordering

any reason why everything in driverkit/GCAdapterDriverKitExtension/Info.plist was reordered?

deReeperJosh, 2026-02-18:
> what ids did you actually change in your build? the diff is not fun to read haha
> 
> or was that simply just reordering
> 
> any reason why everything in driverkit/GCAdapterDriverKitExtension/Info.plist was reordered?

I didn't change or add anything, I think my XCode just automatically reordered it for me