secretkeysio/GCAdapterDriver #30: Not working with MacOS Sequoia

adamschoales, 2025-02-26:
Installed the app, and ran it. Enabled the extension via menubar and then in the security preferences. Loaded up Dolphin: still did not detect the adapter. Disabled the extension via menubar and it said "restart to finish installation". Did so. Same issue. Every time I try enabling it through the menubar I have to re-enable it in system preferences and still Dolphin doesn't actually recognize the adapter. 

This is in MacOS Sequoai 15.3.1

ryanmcgrath, 2025-06-05:
What's it show in `Apple Icon Menu` > `About This Mac` > `System Report` > `USB`? Specifically when you have e.g Slippi open.

To my knowledge nothing should be different in Sequoia, since most people seem fine - but if there's something that needs tweaking I can see what can be done.

greg-wolff, 2025-06-05:
It's a new adapter, I haven't used it before. I'm using GCAdapterDriver v1.6 and it is correctly enabled in Driver Extensions. I've rebooted, but I don't see `GCAdapterDriver` in my `/Library/Extensions` dir.

ryanmcgrath, 2025-06-05:
Well, you wouldn't see it in that directory - it's not a kext.

Are you plugging straight in to your machine (via a small adapter) or through a larger USB hub?

Does regular mainline Dolphin detect it? i.e is this a Slippi-specific issue?

(Bear with me as we debug the million little bits)

greg-wolff, 2025-06-05:
No worries, I'm happy to try and troubleshoot, thanks for the help. The issue seems to be somewhere in the Whisky -> Slippi setup. I tried a few mainline and Development versions of Dolphin for Windows x64, but Whisky doesn't seem to be able to open them. When I download the latest macOS ARM version, the adapter is detected.

ryanmcgrath, 2025-06-05:
...why are they unable to build and distribute a macOS build? If it's a standard fork of Dolphin it should be possible.

If this issue is Whiskey-specific, I'm unlikely to spend too much time digging further. IIRC the author of Whiskey discontinued it, and while I don't know offhand, I would suspect that the issue with the USB connection is due to the binary expecting something from Windows and not getting punted to what macOS does.

(i.e, if both Dolphin's specialized native adapter interface _and_ the standard controller interface aren't working, then there's something else going on there)

ryanmcgrath, 2025-06-05:
I assume [this issue over in the Whiskey repo](https://github.com/Whisky-App/Whisky/issues/592) is also probably related.