secretkeysio/GCAdapterDriver #32: Works on M1 macbook, but ~.5 second input delay?

Danlowe95, 2025-07-04:
https://www.amazon.com/dp/B00L3LQ1FI?ref_=ppx_hzsearch_conn_dt_b_fed_asin_title_2
I'm using this (quite old) variant of the gamecube adapter. Note this is the one without a switch to toggle between WiiU/PC mode.

Macbook m1 Monteray v12.1 

Enabling GCAdapterDriver v1.6 _does_ get the controllers to work, but unfortunately there is a notable input delay making it unusable.

Curious if you've run into this? If I use this same adapter with dolphin on my windows PC I get no input lag. 

Macbook dolphin version is 2506a.

Let me know if there's some known workaround to this issue! It's the last thing stopping me from playing dolphin on my macbook.

ryanmcgrath, 2025-07-04:
Hmm, can you define _input delay_ further? That _looks_ like an old but official adapter, so it should be fine - unless it's some random adapter I've never heard of.

What USB adapter are you using for the A -> C connection? Sometimes those can be finnicky for this kind of thing.

Last resort is you could try overclocking - albeit it's more involved and has some minor tradeoffs: https://github.com/project-slippi/Ishiiruka/wiki/Overclocking-Controllers-on-macOS-Big-Sur,-Monterey,-Ventura,-or-Sonoma

ryanmcgrath, 2025-07-18:
Yeah, so if the _rate_ even with overclocking isn't making a difference, then that does make me think the issue is elsewhere. I do hope you land on something - and if you do, please feel free to log it here so others can benefit from it should they experience it.

If I can help  debug further feel free to ping, but otherwise it sounds like the drivers themselves are working as intended and the issue is something else. :\

ryanmcgrath, 2025-07-21:
Ha, yeah! Makes sense. I carry a small Anker in my bag and it's never let me down. What was your old converter? Might be helpful to list in case others run into the issue.

Going to close this as resolved now - glad you got it working!

nickagliano, 2025-07-21:
The one that didn't work was MOKIN. Looks something like this, although I think I have an older model.

[image]

Danlowe95, 2025-07-21:
Interesting, I wish I had documented what converter I had been using during this test when I had the delay. 

It was one of these two:

Anker USB C Hub (5-in-1), 4K@30Hz HDMI USB Hub:
https://www.amazon.com/dp/B0DKT8BB4M?ref_=ppx_hzsearch_conn_dt_b_fed_asin_title_2

[image]

UGREEN Revodok 6 in 1 USB C Hub:
https://www.amazon.com/dp/B0BW2TLQ8S?ref_=ppx_hzsearch_conn_dt_b_fed_asin_title_1

[image]

I would have guessed it was the Anker, but I can't remember for certain. If I get the chance to test this I'll leave another comment to say if either one works properly. Thanks for investigating!