secretkeysio/GCAdapterDriver #35: Cannot detect mayflash adapter when running slippi

y00j, 2025-08-08:
I am on Sequoia 15.6. I have downloaded the macOS 11.0+ M1 + intel for installation. 

One thing I notice is after I drag the app into my Applications folder, and run it, despite seeing the process running in my activity monitor, I don't see the icon in my menu bar to toggle it on.

ryanmcgrath, 2025-08-08:
Huh, this is... definitely a thing. Googling around it seems that this was a bug in some betas of Sequoia - and I'm not sure if it got fixed?

If you tinker with something like Bartender, is it just hidden by any chance?

sebkkom, 2025-08-13:
Exact same situation here, GCAdapterDriver shows on Activity Monitor but the menu bar icon is nowhere to be found.

> If you tinker with something like Bartender, is it just hidden by any chance?

Nope, even if I remove everything removable from the menu bar, the GCAD icon doesn't show anywhere. :|

ryanmcgrath, 2025-08-13:
Hmmm, very strange but I see what you're both saying. At the moment I cannot find any public-facing API changes in the status bar item for Sequoia+ so... I guess I can look at a release that just does it as a standard NSWindow.

Very unfortunate since I don't really want to have to change a million guides everywhere though.

sebkkom, 2025-08-13:
@ryanmcgrath Maybe just provide a script that does everything temporarily until we figure out what's actually wrong?

sebkkom, 2025-08-13:
Tough luck for us then. Let us know if we can help troubleshoot and/or debug this somehow!

ryanmcgrath, 2025-08-13:
Actually, I'm wondering if the subtle change in Sequoia is that `NSStatusBarItem`s no longer have a default `NSButton`. It looks like I was just taking and customizing the existing button here:

https://github.com/secretkeysio/GCAdapterDriver/blob/b7ab07db4f9f9265621dfeb2811e802424a75a98/driverkit/GCAdapterDriver/AppDelegate.swift#L25

This seems like it might not be triggering and thus there's no button rendering - though the app might be running fine (as you both noted) and just not displaying.

y00j, 2025-08-13:
Hey just an update from me, bartender did fix the problem. You just need to click it a few times and it should show the GCAD icon

[image]

ryanmcgrath, 2025-08-13:
Alright, well that explains some stuff. It looks like there's more padding around... everything... in Sequioa, so maybe it's more aggressive about things getting dropped from the bar?