secretkeysio/GCAdapterDriver #36: DriverKit Implementation: Fixed GetDescriptors() API + Overclocking Support

michaelnorman-au, 2025-09-02:
## Summary
  I've identified and fixed a critical bug in the DriverKit implementation that prevents overclocking functionality for Silicon Macs, and I'm seeking help testing the solution.

  ## Problem Identified
  The current DriverKit implementation fails because `GetDescriptors()` returns NULL descriptors despite successful return codes. This blocks the overclocking feature that works in the kernel extension version.

  ## Solution Implemented
  **Root Cause**: Incorrect parameter passing to `GetDescriptors()` API
  ```cpp
  // OLD (Broken):
  IOUSBStandardEndpointDescriptors *descriptors = NULL;
  pipe->GetDescriptors(descriptors, kIOUSBGetEndpointDescriptorOriginal);

  // NEW (Fixed):
  IOUSBStandardEndpointDescriptors descriptors = {};
  IOUSBStandardEndpointDescriptors *descriptorPtr = &descriptors;
  pipe->GetDescriptors(&descriptorPtr, kIOUSBGetEndpointDescriptorOriginal);

  ```

  Additional Changes:
  - Added #include <USBDriverKit/IOUSBHostInterface.iig> to .iig file
  - Fixed method signature with LOCALONLY attribute

  Testing Blocker

  I can't test this on Apple Silicon due to code signing limitations (personal Apple Developer account can't create DriverKit extensions). The fixed code compiles successfully.

  Request

  Would you be willing to:
  1. Review the technical fix
  2. Build and test it with proper code signing
  3. Consider merging if successful

  This would enable 1000Hz overclocking in the secure DriverKit version without requiring SIP modifications.

  Hardware Tested

  - Nintendo WUP-028 GameCube Controller Adapter
  - Apple Silicon Mac (M1) Pro
  - macOS 15.6.1 

  Happy to provide the modified source code or create a pull request.

ryanmcgrath, 2025-09-03:
Oh wow, this is interesting. You've tested and this works on an Intel Mac?

If so, I am certainly open to patching this - though I might just fold it in to some larger changes I'm making for modern macOS. I'd be happy to provide credit for your work in the README though - let me know what you'd like?

michaelnorman-au, 2025-09-03:
Thanks for getting back to me!

Full disclosure - this is my first contribution to an open source project like this, so I want to be upfront about what I've done vs. what I'm uncertain about.

To clarify, I haven't tested on Intel mac, just my M1 Mac Pro where the fix compiles successfully, but I can't runtime test due to personal developer account limitations with DriverKit signing. The fix addresses what appears to be a parameter passing problem, but I can't be 100% certain it resolves the overclocking issue without proper testing.

And yes would love for this integrated into next patch. For credit, just "USB polling rate fix contributed by michaelnorman-au" in the README works perfectly.

  Let me know if you'd like me to submit a PR or if you prefer to handle the integration directly.

ryanmcgrath, 2025-09-03:
Ah, got it. That makes sense and sounds good to me.

I'm on a vacation trip at the moment but should have time to look at this next week, so I can circle back then. Thanks again for the eyes on this!